Cartographer3D/cartographer3d-plugin #150: Scan to discover bed_mesh and touch boundaries

Jomik, 2025-04-28:
We could create a macro that will run a scan to discover viable mesh_min and mesh_max for the user.

This could be implemented by first scanning a plus to find probable edges, then try to find corners based on this.
Inset the found corners by some safety buffer.
Scan the edge and ensure all measurements are valid and within some tolerance.

GAZ082, 2025-09-26:
I use clips on my bed, some kind of spike detection should be in place as well so the clips are set out of the boundary. The logic should be the same, but instead of checking just for infitine (which i guess is the reading when the sensor does not sense any metal) should check for high values.

The resulting scan should be something like:

[image]

Jomik, 2025-09-26:
> I use clips on my bed, some kind of spike detection should be in place as well so the clips are set out of the boundary. The logic should be the same, but instead of checking just for infinite (which i guess is the reading when the sensor does not sense any metal) should check for high values.

Really good idea, sort of the same idea as #309 .
I would add that as a second iteration, to keep the first feature simple and easy to review / test.