Cartographer3D/cartographer3d-plugin #196: marco TOUCH performs worse than CARTOGRAPHER_TOUCH which in stable version

Bluse-liu50, 2025-06-17:
Because when I used the stable version with lastest version of klipper, printing always failed with "lost connection with mcu scanner", so i tried the alpha version.

I found that the TOUCH works worse than CARTOGRAPHER_TOUCH, appear as too low 0 of z, that the nozzle will rubbing my printing platform.

If i am not misunderstanding, TOUCH should estimate z_offset and save it for current session. It may beacause the alpha remove the fuction of estimate backlash? Just a guess.

Jomik, 2025-06-21:
Hey! No, `TOUCH` is similar to `PROBE`. If you want to update the position, you have to use `TOUCH_HOME`.
Are you part of the closed beta channel in Discord? I do not recognize the username.
Documentation currently lives here https://cartographer-3d.gitbook.io/cartographer-beta-software

closed 2025-06-23 by Jomik

Bluse-liu50, 2025-06-24:
OH, i am not in closed beta... I met the problem of losing connection with scanner after update klipper, so I check your repositories and find this one. I felt sorry for misunderstanding.

And i am not even in the main Channel in discord, cuz I am in China Mainland and we dont use Discord very often.