Cartographer3D/cartographer3d-plugin #265: Feature: make SAVE_CONFIG references clickable.

GAZ082, 2025-08-22:
Some calibration procedures output a text asking the user to do a SAVE_CONFIG, but I have to manually type it instead of clicking on it as in other prompts.

Thanks!

GAZ082, 2025-08-26:
Mainsail. The one that comes to my mind is the SCAN mode calibration. I made a quick search in the code for SAVE_CONFIG, and there is this:
{format_macro('SAVE_CONFIG')},

This format_macro thing is just being used in a test, not in other functions where in the log output there is a SAVE_CONFIG reference. Dont know if this has something to do with it.

Jomik, 2025-08-27:
This is weird. It does work for me in my mainsail.

[image]

GAZ082, 2025-08-27:
Sure Jomik, as soon as i get some printer time i'll run some more scientific tests and report back.

Jomik, 2025-09-01:
Oh... Can't do anything about that.
That's on Klipper https://github.com/Klipper3d/klipper/blob/bab5f8031c8a5dfcfcb20e299da5630ddca0949c/klippy/extras/bed_mesh.py#L1708-L1712
But really, I guess it's on mainsail / fluidd to detect those.
I am not sure our way of doing the clickable links is actually smart too, because it adds html tags to the output.
Sort of debating if we should remove it 🤔 It's only nice because Mainsail and Fluidd both handle it the same way, but some other UIs may not. KlipperScreen it probably looks ugly.

closed 2025-09-03 by Jomik