Cartographer3D/cartographer3d-plugin #311: [Wording] - Change "Acceptable touch combination" to something more generic IF there is no actual "rating" behind "acceptable"

pixel-shock, 2025-09-16:
Hi,

this is not a bug - more a minor "issue" 😉 
I think the wording "acceptable" suggests some rating behind those values - if there is no rating behind it, it would be nice to just change it from

> Acceptable touch combination found

to

> Touch combination found

`Acceptable touch combination found: (0.086962, 0.095712, 0.081962, 0.089462, 0.084462)
maximum 0.095712, minimum 0.081962, range 0.013750,
average 0.087712, median 0.086962,
standard deviation 0.004717, median absolute deviation 0.002500`

PS: Please create labels and types for the issue tracking - I think this will help you later on to filter out 😉

pixel-shock, 2025-09-16:
PPS: If you can do a rating for the touch measurements then it would be even nicer 😉

Jomik, 2025-09-17:
The touches have to satisfy a criteria. Currently seeing what people say with a maximum median absolute deviation of 0.0054.
I do prefer the wording using "acceptable", as it is something that is accepted under some conditions.
If you do more than the 10 touches, then it'll have to pick a combination of 5 from that set that is accepted by our requirements.

Let me know what you think. We can definitely change the wording.
Do note that this is debug output 😄

pixel-shock, 2025-09-17:
Hi,

ah so there is a rating going on in the background - then everything is fine 😉 
Or maybe it would be nice to add some more weighting on the result regarding the output like several thresholds which are then outputting something like "good, bad, excellent, acceptable" etc. ?

Regarding the labels: I can't see any label which I would be able to choose, so i can't say which labels I want to see 😉 
So make the labeling public then would be nice to help you sort those issues a little bit while creating issues?

PS: Touch combinations are imho important to know, so not sure if this should be "verbose" only?

Cheers 🍻

Jomik, 2025-09-19:
That is fair - I think exposing the trigger heights and all the numbers may be a bit much...
And exposing numbers will see people chase perfection, which is definitely not needed.
We have set the threshold so that any accepted touch should be good. Which means, as a user, you should really only care that it was good 😄